canboat/canboatjs #13: analyzerjs produces wind angle in radians while analyzer produced them in degrees

sarfata, 2018-04-19:
As I was writing a test to fix #12, I found another issue:

The angles produced by canboatjs are in radians but analyzer used to print angles in degrees. 

In the example below `analyzer` produced a wind angle of 5.5 but `analyzerjs` says 0.0966 which is the same angle but in radians.

```
echo '$PCDIN,01FD02,00000003,01,00DF00C603FAFFFF*20' |~/work/canboat/rel/darwin-x86_64/analyzer -json
{"timestamp":"1969-12-31-16:00:00,3","prio":0,"src":1,"dst":255,"pgn":130306,"description":"Wind Data","fields":{"SID":0,"Wind Speed":2.23,"Wind Angle":5.5,"Reference":"Apparent"}}
```

And with CanboatJS:
```
echo '$PCDIN,01FD02,00000003,01,00DF00C603FAFFFF*20' |~/work/canboatjs/bin/analyzerjs
{"pgn":130306,"timestamp":3,"src":1,"dst":255,"prio":0,"fields":{"SID":0,"Wind Speed":2.23,"Wind Angle":0.0966,"Reference":"Apparent"},"description":"Wind Data"}
```

I will make a PR for this bug too.

sarfata, 2018-04-19:
(I said "bug" because I assume the original intent was to make canboatjs a drop-in replacement for canboat but of course producing angles in radians might actually be more correct in the NMEA2000/SignalK world...)

sbender9, 2018-04-19:
You should run analyzer with -si

sbender9, 2018-04-19:
canboatjs only supports SI units, I have no plans on changing that.

sarfata, 2018-04-19:
Sounds good! 👍

closed 2018-04-19 by sarfata