canboat/canboatjs #46: Bug: datetime conversion is broken after parseYDWG02 cleanup

neko-koneko, 2019-03-01:
Dear canboat team!

Thank you very much for adding the YDWG-02 support.

I have tested it with real hardware YDWG-02 and found that `parseDate` function of 
`date-fns` package returns an "Invalid Date" in the following line:

https://github.com/canboat/canboatjs/blob/2f88d10f09542dc480a74b9d2896ea6d831e89b7/lib/fromPgn.js#L283

It seems that you call `parseDate` function with the arguments list corresponding to a new v2.0.0 `date-fns` version, while dependencies section of a package.json requires an old v1.30.1 version 

https://github.com/webmasterkai/canboatjs/blob/79e33b4879e720dcfc23bb1d846797d55be780bd/package.json#L43

 `date-fns` v1.30.1 version will not work as it's `parseDate` function requires a different set of arguments

https://github.com/date-fns/date-fns/blob/0ccdbbf1c4d88afcfed480f25186a7463b3b9726/src/parse/index.js#L193

Previous version of `parseYDWG02` function, where the date is parsed without a `date-fns` library, works fine
https://github.com/canboat/canboatjs/blob/25f35c9e4e8daebbcc04d4c067f6ed4c0e16e0a9/lib/fromPgn.js#L265

I have updated the `date-fns` package to a latest version 2.0.0-alpha.27 and find that it works if you change the L283 of `fromPgn.js` to the following:
`pgn.timestamp = parseDate(time, 'HH:mm:ss.SSS', new Date()).toISOString()`

Please, consider to either fix dependencies or revert back to the original timedate parsing algorithm.

Best regards,
Alexey Korolev,
support engineer,
Yacht Devices Ltd

sbender9, 2019-03-04:
Thanks Alexey. 

Any chance you guys can send me the hardware? It would be nice to be able to test with the real thing. 

I will make your recommended changes ASAP.

closed 2019-03-04 by sbender9

sbender9, 2019-03-04:
this has been releases in canboatjs v1.3.1

neko-koneko, 2019-03-13:
Dear Scott!Thank you very much again for adding the support of our gateway!You have asked on a github: >Any chance you guys can send me the hardware? It would be nice to be able to test with the real thing.We have decided that we can send you a YDWG-02 unit.Please, tell us which NMEA 2000 connector is required:- DeviceNet NMEA 2000 Micro Maleor- Raymarine SeaTalk NG.Please, tell us the postal address we should send the device to, as well as recipient full name. Best regards,Alexey Korolev,support engineer,Yacht Devices Ltdwww.yachtd.com

sbender9, 2019-03-13:
I sent the info in an email to neko-koneko@yandex.ru

thanks!

PR #47 (2019-03-04): fix: YDWG date parse error
parseDate function with the arguments list corresponding to a new v2.0.0 date-fns version, while dependencies section of a package.json requires an old v1.30.1 version
updated the package dependencies and updated the ydwg parse to use toISOString

Closes #46
commit 4d5a9cb5 (2019-03-04): fix: YDWG date parse error (#47)

parseDate function with the arguments list corresponding to a new v2.0.0 date-fns version, while dependencies section of a package.json requires an old v1.30.1 version
updated the package dependencies and updated the ydwg parse to use toISOString

Closes #46