canboat/canboatjs #91: License type of canboatjs and @canboat/pgns

pjp8w, 2019-11-20:
The license details of the canboatjs repo is Apache 2.0.
When I look at the package.json for @canboat/pgns (https://github.com/canboat/canboat/blob/master/analyzer/package.json) it also indicates Apache 2.0.

However, when I look at pgns.json there is a License element indicating GPL v3. 
https://github.com/canboat/canboat/blob/master/analyzer/pgns.json

Can someone confirm what exactly the full license type for canboatjs is please. The reason for asking is GPL is a copyleft license and I wish to use canboatjs from an Apache 2.0 terms perspective. Perhaps the pgns.json file has a legacy license element that needs to be removed?

webmasterkai, 2019-11-20:
The PGNS are from the canboat/canboat repo that is licensed GNU GPL 3. My understanding is the javascript code here is Apache 2 but not the PGN file.

If you are creating closed source code it's probably better to purchase the definition files/licenses directly from NMEA and/or SAE anyway.

pjp8w, 2019-11-20:
Thanks for the response. We are looking at closed source, yes, and have actually purchased the definition files/licenses from NMEA already. The definitions are a PDF document so cannot be used programmatically directly. 
Ideally we'd be hoping to use the canboatjs codebase for decoding but obviously with the GPL license of the pgn file we would be blocked from going down this route. 
It's my understanding that using canboatjs would make any third party software GPL given the viral nature of the copyleft license of the PGN file.

webmasterkai, 2019-11-20:
Curiously https://www.npmjs.com/package/@canboat/pgns lists the license as Apache 2. Perhaps that's a mistake?

I think you'll run into problems when you notice something is wrong or outdated in the canboat/canboat repo that creates the PGNs. Once you purchase the definition file you are no longer allowed to contribute/publish details from the spec to the public repo. NMEA has a Microsoft Access database file called “NMEA Network Message.mdb” (mentioned on pg 16). Have you asked them for a copy of it?

After you fix a mistake you'll be maintaining a private fork of the canboat pgns and wanting to use that instead of what is included in the canboatjs repo. Canboatjs doesn't have an easy way of bringing your own definition files... So then you end up with a fork of canboatjs that allows BYO PGN info...

Also, you might find that the problem is in a field that's used on many PGNs but there isn't an easy way to find all instances of a particular DD# and copy paste so you'll probably end up building a database or at least splitting out the fields so fixes/updates can be used in every PGN that uses it... 

Feel free to send me an email if you want to exchange thoughts. I've been down this road and would be happy to help if I can.

pjp8w, 2019-11-20:
> Curiously https://www.npmjs.com/package/@canboat/pgns lists the license as Apache 2. Perhaps that's a mistake?

That was essentially my original question. The package.json file says Apache 2.0 but the PGN file has forever had GPL - perhaps a legacy thing that was never updated. 

>NMEA has a Microsoft Access database file called “NMEA Network Message.mdb” (mentioned on pg 16). Have you asked them for a copy of it?

Yes actually. The response from NMEA was "NMEA does not provide the actual .mdb file.  As stated on the NMEA website, you receive an PDF report." 

> So then you end up with a fork of canboatjs that allows BYO PGN info...

See that's the thing... If the pgns.json file is GPL, you can't just fork it and work away on your own. Which comes back to the question, "What is the real license of that file?". Do you happen to know who to ask that of?

sbender9, 2019-11-20:
The pgns file is generated via the canboat project which is GPL v3: https://github.com/canboat/canboat/blob/master/LICENSE

I will update the package.json to reflect that in the next release.

I probably should have made canboatjs GPL since it uses resources from canboat.

webmasterkai, 2019-11-20:
The json file with the pgn info is generated based on the code in the canboat/canboat that has https://github.com/canboat/canboat/blob/master/LICENSE
I think @sbender9 put Apache as the license in the package.json file by mistake. The person to talk to would be keesverruijt.

webmasterkai, 2019-11-20:
I’ve turned the Appendix A&B PDF into a database for my own use. I've asked NMEA if I can sell a copy of the database I created to someone that has a valid license to Appendix A&B.

sbender9, 2022-01-14:
canboat and canboatjs are now Apache 2.0

closed 2022-01-14 by sbender9