canboat/canboatjs #110: NULL values

fabdrol, 2020-04-07:
NMEA 2000 sends 0xFF in a field for "no value". From canboatjs I get "-1" when I would expect "no value" (null?)

For instance, engine:

```
1|canboat-listener  | 13:46:53.973: {
1|canboat-listener  | 13:46:53.973:   "canId": 233963937,
1|canboat-listener  | 13:46:53.973:   "prio": 3,
1|canboat-listener  | 13:46:53.973:   "src": 161,
1|canboat-listener  | 13:46:53.973:   "dst": 255,
1|canboat-listener  | 13:46:53.973:   "pgn": 127489,
1|canboat-listener  | 13:46:53.973:   "timestamp": "2020-04-07T13:46:53.973Z",
1|canboat-listener  | 13:46:53.973:   "input": [],
1|canboat-listener  | 13:46:53.973:   "fields": {
1|canboat-listener  | 13:46:53.973:     "Instance": "Dual Engine Starboard",
1|canboat-listener  | 13:46:53.973:     "Oil pressure": 0,
1|canboat-listener  | 13:46:53.973:     "Temperature": 293.15,
1|canboat-listener  | 13:46:53.973:     "Alternator Potential": 12,
1|canboat-listener  | 13:46:53.973:     "Fuel Rate": 0,
1|canboat-listener  | 13:46:53.973:     "Total Engine hours": 0,
1|canboat-listener  | 13:46:53.973:     "Fuel Pressure": 0,
1|canboat-listener  | 13:46:53.973:     "Discrete Status 1": [],
1|canboat-listener  | 13:46:53.973:     "Discrete Status 2": [
1|canboat-listener  | 13:46:53.973:       null,
1|canboat-listener  | 13:46:53.973:       null,
1|canboat-listener  | 13:46:53.973:       null,
1|canboat-listener  | 13:46:53.973:       null,
1|canboat-listener  | 13:46:53.973:       null,
1|canboat-listener  | 13:46:53.973:       null,
1|canboat-listener  | 13:46:53.973:       null,
1|canboat-listener  | 13:46:53.973:       null
1|canboat-listener  | 13:46:53.973:     ],
1|canboat-listener  | 13:46:53.973:     "Percent Engine Load": -1,
1|canboat-listener  | 13:46:53.973:     "Percent Engine Torque": -1
1|canboat-listener  | 13:46:53.973:   },
1|canboat-listener  | 13:46:53.973:   "description": "Engine Parameters, Dynamic"
1|canboat-listener  | 13:46:53.973: }
```

fabdrol, 2020-04-07:
What would be the correct behaviour? `pgns.json` states that `Percent Engine Load` and `Percent Engine Torque` are percentages in the range 0-100%. -1 is not in that range.

sbender9, 2020-04-07:
Yeah, it should be null. I’ll take a look...

sbender9, 2020-04-07:
Can you give me the raw input?

sbender9, 2020-04-07:
So that's actually signed number +/- 124% . So null value should be 0x7f.

An I've confirmed that in the test data, some of which is from my boat:

2017-04-15T15:18:51.313Z,2,127489,17,255,26,00,f5,0f,ff,ff,8f,7f,ff,7f,ff,7f,64,af,0a,00,ff,ff,ff,ff,7f,00,00,00,00,7f,7f

fabdrol, 2020-04-07:
@sbender9 I don't actually have the raw input, but I can get it. Stand by....

fabdrol, 2020-04-07:
```
2020-04-07-14:47:39.916 3 161 255 127489 Engine Parameters, Dynamic:  Instance = Dual Engine Starboard; Oil pressure = 2.800 bar (193.1 PSI); Temperature = 96.00 C (204.8 F); Alternator Potential = 10.50 V; Fuel Rate = 1.3 L/h; Total Engine hours = 187200 s; Fuel Pressure = 30 kPa; Discrete Status 1 =; Discrete Status 2 = "256","512","1024","2048","4096","8192","16384","32768"; Percent Engine Load = 78; Percent Engine Torque = -1
2020-04-07-14:47:39.916 3 161 255 127489 : 03 00 00 00 ff 4e ff ff
```

sbender9, 2020-04-07:
So that’s bad data, it’s not following the spec.

fabdrol, 2020-04-07:
What's going wrong?

sbender9, 2020-04-07:
Should be 0x7f for null because it is a signed number.

fabdrol, 2020-04-07:
Why is it signed, btw? It's a 0-100 integer?

sbender9, 2020-04-07:
I can’t explain why, but that’s how it is defined in pgns and all the sample data I have uses 0x7f for null. 

What’s this coming from?

fabdrol, 2020-04-07:
@sbender9 a simulator, so easy to adjust. I reckon this can be closed :)

closed 2020-04-07 by fabdrol