canboat/canboatjs #117: parser fromPgn returns SOG that is not expected should be 0

okbayou, 2020-05-27:
After creating a parser like the one in readme.md most pgns worked OK except for: 129026 COG & SOG, Rapid Update
SOG is apparently reading data from a reserved field.  The data from a connected GPS is 0 yet it returns a much higher value.
data verified with  Actisense NGT-1 connected to a PC running NMEA 2000 reader which showed SOG = 00.
raw data: 23:10:04.070 R 09f80204 ff fc 55 b7 0 0 ff ff
Fields returned: {"COG Reference":"True","COG":4.6933,"SOG":652.8}

test code:
var test2 = "23:34:50.784 R 09f80204 ff fc b4 4b 0 0 ff ff";
const FromPgn = canboat.FromPgn;
const parser = new FromPgn();
const json = parser.parseString(test2);
if ( json ) {
  msg.payload = json;
}
return msg;
The above code returned the following:
msg.payload = 
{"canId":167248388,
"prio":2,
"src":4,
"dst":255,
"pgn":129026
,"direction":"R",
"time":"23:34:50.784",
"input":
["23:34:50.784 R 09f80204 ff fc b4 4b 0 0 ff ff"],
"fields":
   {"COG Reference":"True",
    "COG":1.938,
    "SOG":652.8},
   "description":"COG & SOG, Rapid Update",
   "timestamp":"2020-05-27T23:38:07.119Z"}

sbender9, 2020-05-28:
Your input is messed up. The zeros should be two characters, `00`:

`23:34:50.784 R 09f80204 ff fc b4 4b 00 00 ff ff`

okbayou, 2020-05-28:
OK, thanks!

closed 2020-05-28 by okbayou