canboat/canboatjs #145: Actisense NGT-1 stream dies

preeve9534, 2020-12-10:
This issue was initially raised at [https://github.com/SignalK/signalk-server/issues/1157[(https://github.com/SignalK/signalk-server/issues/1157).

In summary:

Actisense NGT-1 (latest firmware)
canboatjs@1.18.04
signalk-server@1.36.0

After several hours (between 4 and 48) the data stream from canboatjs locks up and the NGT-1 is effectively dead and can usually only be recovered by a power reset.  The host server logs show no relevant error messages.

The problem is not evident in SignalK when actisense-serial is used instead of canboatjs, hence I suspect the issue is with this application rather than the host context.

sbender9, 2023-05-28:
Did you ever figure what the issue was?

preeve9534, 2023-05-29:
Hello Scott. Well not exactly.

I sent the NGT1 back to Actisense who could find no issues with the device. I then decided to switch my Signal K instalation onto CerboGX that Victron gave me. Same Signal K installation, but the problem disappeared.

Since then, I have used the NGT1 with a fresh Signal K installation and the intermittent lock ups are still there. canboatjs just dies - no errors in the logs, only remedy is a reboot of the host PC hardware. Reset of the NGT1 by disconnecting NMEA bus has no effect; reboot of signalk-server has no effect.

Sorry not to be able to offer more help.

Paul

closed 2024-08-24 by sbender9