canboat/canboatjs #157: Canboatjs device not recognized by Garmin plotter

danka74, 2021-03-27:
After an update of openplotter 2 earlier this year my plotter (Garmin 721xs) no longer received NMEA 2000 data from my Raspberry with PICAN-M NMEA 2000 interface. Sorry, I don't have details of previous version as I didn't expect to have to debug this issue...

The Raspberry/openplotter shows up briefly (2 seconds) on the device list of the Garmin plotter but disappears quickly. Showing logs when swithcing the plotter on (the plotter seems to request info and the Raspberry is sending, but the plotter sends nothing back).

Any thoughts?

Current version info:
canboatjs@1.1.10 
signalk-server@1.38.1 

Output from debug of canboatjs:candevice switching on 721xs:
```
Mar 27 15:02:09 2021-03-27T14:02:09.478Z canboatjs:candevice handleISORequest {"canId":418053888,"prio":6,"src":0,"dst":255,"pgn":59904,"timestamp":"2021-03-27T14:02:09.474Z","fields":{"PGN":60928},"description":"ISO Request"}
Mar 27 15:02:09 2021-03-27T14:02:09.480Z canboatjs:candevice Sending PGN {"pgn":60928,"dst":0,"Unique Number":191602,"Manufacturer Code":999,"Device Function":130,"Device Class":25,"Device Instance Lower":0,"Device Instance Upper":0,"System Instance":0,"Industry Group":4,"Reserved1":1,"Reserved2":2,"src":100}
Mar 27 15:02:11 2021-03-27T14:02:11.933Z canboatjs:candevice handleISORequest {"canId":418014208,"prio":6,"src":0,"dst":100,"pgn":59904,"timestamp":"2021-03-27T14:02:11.932Z","fields":{"PGN":126996},"description":"ISO Request"}
Mar 27 15:02:11 2021-03-27T14:02:11.934Z canboatjs:candevice Sending product info..
Mar 27 15:02:11 2021-03-27T14:02:11.935Z canboatjs:candevice Sending PGN {"pgn":126996,"dst":255,"NMEA 2000 Version":1300,"Product Code":667,"Model ID":"Signal K","Software Version Code":"1.0","Model Version":"canbusjs","Model Serial Code":"123456","Certification Level":0,"Load Equivalency":1,"src":100}
Mar 27 15:02:12 2021-03-27T14:02:12.433Z canboatjs:candevice handleISORequest {"canId":418014208,"prio":6,"src":0,"dst":100,"pgn":59904,"timestamp":"2021-03-27T14:02:12.433Z","fields":{"PGN":126996},"description":"ISO Request"}
Mar 27 15:02:12 2021-03-27T14:02:12.434Z canboatjs:candevice Sending product info..
Mar 27 15:02:12 2021-03-27T14:02:12.435Z canboatjs:candevice Sending PGN {"pgn":126996,"dst":255,"NMEA 2000 Version":1300,"Product Code":667,"Model ID":"Signal K","Software Version Code":"1.0","Model Version":"canbusjs","Model Serial Code":"123456","Certification Level":0,"Load Equivalency":1,"src":100,"prio":3}
Mar 27 15:02:12 2021-03-27T14:02:12.934Z canboatjs:candevice handleISORequest {"canId":418014208,"prio":6,"src":0,"dst":100,"pgn":59904,"timestamp":"2021-03-27T14:02:12.934Z","fields":{"PGN":126996},"description":"ISO Request"}
Mar 27 15:02:12 2021-03-27T14:02:12.935Z canboatjs:candevice Sending product info..
Mar 27 15:02:12 2021-03-27T14:02:12.936Z canboatjs:candevice Sending PGN {"pgn":126996,"dst":255,"NMEA 2000 Version":1300,"Product Code":667,"Model ID":"Signal K","Software Version Code":"1.0","Model Version":"canbusjs","Model Serial Code":"123456","Certification Level":0,"Load Equivalency":1,"src":100,"prio":3}
Mar 27 15:02:13 2021-03-27T14:02:13.435Z canboatjs:candevice handleISORequest {"canId":418014208,"prio":6,"src":0,"dst":100,"pgn":59904,"timestamp":"2021-03-27T14:02:13.435Z","fields":{"PGN":126996},"description":"ISO Request"}
Mar 27 15:02:13 2021-03-27T14:02:13.436Z canboatjs:candevice Sending product info..
Mar 27 15:02:13 2021-03-27T14:02:13.437Z canboatjs:candevice Sending PGN {"pgn":126996,"dst":255,"NMEA 2000 Version":1300,"Product Code":667,"Model ID":"Signal K","Software Version Code":"1.0","Model Version":"canbusjs","Model Serial Code":"123456","Certification Level":0,"Load Equivalency":1,"src":100,"prio":3}
Mar 27 15:02:13 2021-03-27T14:02:13.936Z canboatjs:candevice handleISORequest {"canId":418014208,"prio":6,"src":0,"dst":100,"pgn":59904,"timestamp":"2021-03-27T14:02:13.936Z","fields":{"PGN":126996},"description":"ISO Request"}
Mar 27 15:02:13 2021-03-27T14:02:13.938Z canboatjs:candevice Sending product info..
Mar 27 15:02:13 2021-03-27T14:02:13.940Z canboatjs:candevice Sending PGN {"pgn":126996,"dst":255,"NMEA 2000 Version":1300,"Product Code":667,"Model ID":"Signal K","Software Version Code":"1.0","Model Version":"canbusjs","Model Serial Code":"123456","Certification Level":0,"Load Equivalency":1,"src":100,"prio":3}
```
vs. when I have a second plotter (Garmin 62cv, receives NMEA 2000 data from Raspberry just fine!):
```
Mar 27 15:04:50 2021-03-27T14:04:50.784Z canboatjs:candevice registering device 1
Mar 27 15:04:51 2021-03-27T14:04:51.043Z canboatjs:candevice handleISORequest {"canId":418053889,"prio":6,"src":1,"dst":255,"pgn":59904,"timestamp":"2021-03-27T14:04:51.042Z","fields":{"PGN":60928},"description":"ISO Request"}
Mar 27 15:04:51 2021-03-27T14:04:51.044Z canboatjs:candevice Sending PGN {"pgn":60928,"dst":1,"Unique Number":191602,"Manufacturer Code":999,"Device Function":130,"Device Class":25,"Device Instance Lower":0,"Device Instance Upper":0,"System Instance":0,"Industry Group":4,"Reserved1":1,"Reserved2":2,"src":100}
Mar 27 15:04:52 2021-03-27T14:04:52.225Z canboatjs:candevice got production information {"canId":435164161,"prio":6,"src":1,"dst":255,"pgn":126996,"timestamp":"2021-03-27T14:04:52.224Z","input":[],"fields":{"NMEA 2000 Version":2100,"Product Code":25165,"Model ID":"ECHOMAP Plus 62cv\u0000","Software Version Code":"13.30\u0000","Model Version":"1.0\u0000","Model Serial Code":"3963789777\u0000","Certification Level":2,"Load Equivalency":1},"description":"Product Information"}
Mar 27 15:04:52 2021-03-27T14:04:52.571Z canboatjs:candevice handleISORequest {"canId":418014209,"prio":6,"src":1,"dst":100,"pgn":59904,"timestamp":"2021-03-27T14:04:52.571Z","fields":{"PGN":126996},"description":"ISO Request"}
Mar 27 15:04:52 2021-03-27T14:04:52.572Z canboatjs:candevice Sending product info..
Mar 27 15:04:52 2021-03-27T14:04:52.573Z canboatjs:candevice Sending PGN {"pgn":126996,"dst":255,"NMEA 2000 Version":1300,"Product Code":667,"Model ID":"Signal K","Software Version Code":"1.0","Model Version":"canbusjs","Model Serial Code":"123456","Certification Level":0,"Load Equivalency":1,"src":100,"prio":3}
Mar 27 15:04:53 2021-03-27T14:04:53.068Z canboatjs:candevice handleISORequest {"canId":418014209,"prio":6,"src":1,"dst":100,"pgn":59904,"timestamp":"2021-03-27T14:04:53.067Z","fields":{"PGN":126996},"description":"ISO Request"}
Mar 27 15:04:53 2021-03-27T14:04:53.069Z canboatjs:candevice Sending product info..
Mar 27 15:04:53 2021-03-27T14:04:53.070Z canboatjs:candevice Sending PGN {"pgn":126996,"dst":255,"NMEA 2000 Version":1300,"Product Code":667,"Model ID":"Signal K","Software Version Code":"1.0","Model Version":"canbusjs","Model Serial Code":"123456","Certification Level":0,"Load Equivalency":1,"src":100,"prio":3}
Mar 27 15:04:53 2021-03-27T14:04:53.569Z canboatjs:candevice handleISORequest {"canId":418014209,"prio":6,"src":1,"dst":100,"pgn":59904,"timestamp":"2021-03-27T14:04:53.569Z","fields":{"PGN":126996},"description":"ISO Request"}
Mar 27 15:04:53 2021-03-27T14:04:53.570Z canboatjs:candevice Sending product info..
Mar 27 15:04:53 2021-03-27T14:04:53.570Z canboatjs:candevice Sending PGN {"pgn":126996,"dst":255,"NMEA 2000 Version":1300,"Product Code":667,"Model ID":"Signal K","Software Version Code":"1.0","Model Version":"canbusjs","Model Serial Code":"123456","Certification Level":0,"Load Equivalency":1,"src":100,"prio":3}
Mar 27 15:04:54 2021-03-27T14:04:54.094Z canboatjs:candevice handleISORequest {"canId":418014209,"prio":6,"src":1,"dst":100,"pgn":59904,"timestamp":"2021-03-27T14:04:54.094Z","fields":{"PGN":126996},"description":"ISO Request"}
Mar 27 15:04:54 2021-03-27T14:04:54.095Z canboatjs:candevice Sending product info..
Mar 27 15:04:54 2021-03-27T14:04:54.097Z canboatjs:candevice Sending PGN {"pgn":126996,"dst":255,"NMEA 2000 Version":1300,"Product Code":667,"Model ID":"Signal K","Software Version Code":"1.0","Model Version":"canbusjs","Model Serial Code":"123456","Certification Level":0,"Load Equivalency":1,"src":100,"prio":3}
Mar 27 15:04:54 2021-03-27T14:04:54.595Z canboatjs:candevice handleISORequest {"canId":418014209,"prio":6,"src":1,"dst":100,"pgn":59904,"timestamp":"2021-03-27T14:04:54.595Z","fields":{"PGN":126996},"description":"ISO Request"}
Mar 27 15:04:54 2021-03-27T14:04:54.597Z canboatjs:candevice Sending product info..
[... cut ...] evel":0,"Load Equivalency":1,"src":100,"prio":3}
Mar 27 15:04:55 2021-03-27T14:04:55.021Z canboatjs:candevice got production information {"canId":435164160,"prio":6,"src":0,"dst":255,"pgn":126996,"timestamp":"2021-03-27T14:04:55.020Z","input":[],"fields":{"NMEA 2000 Version":1300,"Product Code":22811,"Model ID":"GPSMAP 721xs","Software Version Code":"6.80","Model Version":"1.0","Model Serial Code":"3884074848","Certification Level":1,"Load Equivalency":2},"description":"Product Information"}
```

fymmot, 2022-08-05:
I think I may have the same problem. Did you find a solution @danka74 ?

My Garmin 527xs plotter is not able to receive AIS targets from the RPi over NMEA2000. The RPi is not listed among network devices. I too get a connection momentarily after booting the plotter, with 2-3 AIS targets displayed briefly, but they soon disappear. 

My system:

RPi 4 with Pican-M hat
OpenPlotter 3 beta 
Garmin 527xs plotter
SignalK version 1.44.0
canboatjs

danka74, 2022-08-05:
@fymmot sorry, I did not, but I'm "lucky" that I got one working plotter (Garmin 62cv). As a matter of fact, I haven't tried this season, but might do later. It's in the boat in its box.

fymmot, 2022-08-05:
Ah that's too bad! But at least I'm a little closer to understanding what the cause might be. Your 721xs and my 527xs are from the same Garmin series, it doesn't feel like a coincidence.

fymmot, 2023-07-24:
This issue is still affecting my installation and preventing my Garmin 527xs plotter from receiving any data over N2K. 

Would anyone be willing to look at it for me? I would be very grateful!

dpazz, 2023-07-29:
I incurred in the same issue attempting to connect a Garmin GMI20 display connected via NMEA2K to a canboatjs device (driven by SignalK hosted in a rock-4se SBC -an alternative to RaspberryPI Single Board Computer I bought during RPI4 shortage - ). In the "NMEA2000" config page of the Garnmin display the canboatjs device appears as   "unknown device". Going into the device menu the Garmini repeatss "device unknown" and states "seral number= invalid"   . Very rarely I can see the GPS position received from Garmin, but it is quicly "discoonected" and recorded in the event register as "lost GPS connecction". I investigated furtherly and I found in the trace obtained typing in a linux terminal the command pipe "candump can0 | candump2analyzer | analyzer" that the CANbus Address Claim phase is repeated many times. I suppose that  Garmin dev is attempting to request many times the product info of the canboatjs counterpart, obtaining unsatisfactory replies (in its "opinion"). I suppose that this behavior is depending on the firmware release of Garmin devices that is doing a more strict control on "compatible" N2K counterparts (this should explain why "older" Garmin or older fw release are less choosy). I was wondering if it is possible to "hack" the canboatjs in order to fool the choosy Garmin, e.g. hardcoding a different "serial" sent inside the product info PGN. The actual serial appears in the trace as "123456". Hereafter there are some pictures taken from GMI20 (sorry for the italian)
[image]
[image]
[image]
[image]
[image]
[202307291232 - candump can0 %7C candump2analyzer %7C analyzer %7C grep Claim- trace.txt](https://github.com/canboat/canboatjs/files/12206910/202307291232.-.candump.can0.7C.candump2analyzer.7C.analyzer.7C.grep.Claim-.trace.txt)
[20230729-analyzer_debug_output.txt](https://github.com/canboat/canboatjs/files/12206911/20230729-analyzer_debug_output.txt)

sbender9, 2023-07-29:
So far I have not been able to figure what Garmin doesn't like here. I plan on playing more with this next week...

dpazz, 2023-07-29:
I suppose looking at the analyzer output that Garmin dislike the response canboatjs give to its product info query. I tried to play with no joy 'hacking' serial number in candevice.js. Now I've to dismount my test plant. Maybe in three to four days I'll we back ...

sbender9, 2024-08-24:
Closing because of inactivity. I think some recent changes may have fixed it. Please let me know...

closed 2024-08-24 by sbender9

rhbvkleef, 2025-10-30:
I am currently having this issue (or a very similar one) with v 3.12.1. My suspicion is that the dst field is set to 255, and not to the address of the requester.