canboat/canboatjs #162: Incorrect instances being generated

nmbath, 2021-07-29:
I have found an issue whereby malformed packets seem to be processed and generate fake instance.

There is a thread on slack (https://signalk-dev.slack.com/archives/C03F1MKQG/p1627240871182300), where I have reported an issue.

I thought this might have been unique to my install and PGN127506, but a second user has reported this happening with PGN127497. 

In the below debug, there is no way the Victron device at CAN address 41 (0x29) has a battery instance of 56

Jul 26 17:35:53 localhost signalk-server[1369823]: 2021-07-26T17:35:53.731Z canboatjs:fromPgn parsed pgn {"canId":435294761,"prio":6,"src":41,"dst":255,"pgn":127506,"timestamp":"2021-07-26T17:35:53.730Z","fields":{"SID":97,"State of Health":144,"Time Remaining":65281},"description":"DC Detailed Status"}
Jul 26 17:35:56 localhost signalk-server[1369823]: 2021-07-26T17:35:56.730Z canboatjs:fromPgn PGN 127506 malformed packet for 41 received; expected 1 but got 161
Jul 26 17:35:56 localhost signalk-server[1369823]: 2021-07-26T17:35:56.730Z canboatjs:fromPgn parsed pgn {"canId":435294761,"prio":6,"src":41,"dst":255,"pgn":127506,"timestamp":"2021-07-26T17:35:56.730Z","fields":{"SID":161,"Instance":56,"State of Health":144,"Time Remaining":65281},"description":"DC Detailed Status"}
Jul 26 17:37:05 localhost signalk-server[1369823]: 2021-07-26T17:37:05.809Z canboatjs:fromPgn PGN 127506 malformed packet for 228 received; expected 1 but got 161
Jul 26 17:37:08 localhost signalk-server[1369823]: 2021-07-26T17:37:08.812Z canboatjs:fromPgn PGN 127506 malformed packet for 228 received; expected 1 but got 225

Debugging this as I create the issue, the below screen shot shows another point where the data is invalid.

[image]

The syslog messages from around this time, with canboatjs:fromPgn debugging turned on... gives

Jul 29 11:00:49 localhost signalk-server[2381537]: 2021-07-29T11:00:37.339Z canboatjs:fromPgn parsed pgn {"canId":435294948,"prio":6,"src":228,"dst":255,"pgn":127506,"timestamp":"2021-07-29T11:00:37.339Z","input":[],"fields":{"SID":24,"Instance":5,"DC Type":"Battery"},"description":"DC Detailed Status"}
Jul 29 11:00:49 localhost signalk-server[2381537]: 2021-07-29T11:00:37.851Z canboatjs:fromPgn parsed pgn {"canId":435294944,"prio":6,"src":224,"dst":255,"pgn":127506,"timestamp":"2021-07-29T11:00:37.851Z","input":[],"fields":{"SID":34,"Instance":2,"DC Type":"Battery","State of Charge":100},"description":"DC Detailed Status"}
Jul 29 11:00:49 localhost signalk-server[2381537]: 2021-07-29T11:00:37.990Z canboatjs:fromPgn PGN 127506 malformed packet for 227 received; expected 1 but got 129
Jul 29 11:00:49 localhost signalk-server[2381537]: 2021-07-29T11:00:37.991Z canboatjs:fromPgn parsed pgn {"canId":435294947,"prio":6,"src":227,"dst":255,"pgn":127506,"timestamp":"2021-07-29T11:00:37.990Z","fields":{"SID":129},"description":"DC Detailed Status"}
Jul 29 11:00:49 localhost signalk-server[2381537]: 2021-07-29T11:00:38.121Z canboatjs:fromPgn parsed pgn {"canId":435294758,"prio":6,"src":38,"dst":255,"pgn":127506,"timestamp":"2021-07-29T11:00:38.121Z","input":[],"fields":{"SID":139,"Instance":0,"DC Type":"Battery"},"description":"DC Detailed Status"}
Jul 29 11:00:49 localhost signalk-server[2381537]: 2021-07-29T11:00:38.124Z canboatjs:fromPgn parsed pgn {"canId":435294758,"prio":6,"src":38,"dst":255,"pgn":127506,"timestamp":"2021-07-29T11:00:38.124Z","input":[],"fields":{"SID":139,"Instance":1,"DC Type":"Solar Cell"},"description":"DC Detailed Status"}
Jul 29 11:00:49 localhost signalk-server[2381537]: 2021-07-29T11:00:38.580Z canboatjs:fromPgn parsed pgn {"canId":435294761,"prio":6,"src":41,"dst":255,"pgn":127506,"timestamp":"2021-07-29T11:00:38.580Z","input":[],"fields":{"SID":235,"Instance":0,"DC Type":"Battery","State of Charge":100},"description":"DC Detailed Status"}

and candump 

 (2021-07-29 11:00:37.839502)  can1  RX - -  19F212E0   [8]  C0 09 22 02 00 64 FF FF
 (2021-07-29 11:00:37.840143)  can1  RX - -  19F212E0   [8]  C1 FF FF FF FF FF FF FF
 (2021-07-29 11:00:37.988912)  can1  RX - -  19F212E3   [8]  81 FF FF FF FF FF FF FF
 (2021-07-29 11:00:38.119447)  can1  RX - -  19F21226   [8]  20 0B 8B 00 00 FF FF FF
 (2021-07-29 11:00:38.120107)  can1  RX - -  19F21226   [8]  21 FF FF FF FF FF FF FF
 (2021-07-29 11:00:38.121154)  can1  RX - -  19F21226   [8]  20 0B 8B 01 03 FF FF FF

If I push these messages through canboat analyser I get 

INFO 2021-07-29T11:11:25.327Z [analyzer] New PGN 127506 for device 224 (heap 11154 bytes)
2021-07-29-11:11:25.327 6 224 255 127506 DC Detailed Status:  SID = 34; Instance = 2; DC Type = Battery; State of Charge = 100; State of Health = Unknown; Time Remaining = Unknown; Ripple Voltage = Unknown
INFO 2021-07-29T11:11:25.327Z [analyzer] New PGN 127506 for device 227 (heap 22308 bytes)
ERROR 2021-07-29T11:11:25.327Z [analyzer] PGN 127506 malformed packet from 227 received; expected 1 but got 129
INFO 2021-07-29T11:11:25.327Z [analyzer] New PGN 127506 for device 38 (heap 33462 bytes)
2021-07-29-11:11:25.327 6  38 255 127506 DC Detailed Status:  SID = 139; Instance = 0; DC Type = Battery; State of Charge = Unknown; State of Health = Unknown; Time Remaining = Unknown; Ripple Voltage = Unknown; Amp Hours = Unknown

and through canboatanalyserjs

{"canId":435294944,"prio":6,"src":224,"dst":255,"pgn":127506,"bus":"can1","input":["can1  19F212E0   [8]  C0 09 22 02 00 64 FF FF","can1  19F212E0   [8]  C1 FF FF FF FF FF FF FF","can1  19F212E0   [8]  C1 FF FF FF FF FF FF FF"],"fields":{"SID":34,"Instance":2,"DC Type":"Battery","State of Charge":100},"description":"DC Detailed Status","timestamp":"2021-07-29T11:12:22.153Z"}
{"canId":435294947,"prio":6,"src":227,"dst":255,"pgn":127506,"bus":"can1","fields":{"SID":129},"description":"DC Detailed Status","timestamp":"2021-07-29T11:12:22.160Z"}
{"canId":435294758,"prio":6,"src":38,"dst":255,"pgn":127506,"bus":"can1","input":["can1  19F21226   [8]  20 0B 8B 00 00 FF FF FF","can1  19F21226   [8]  21 FF FF FF FF FF FF FF","can1  19F21226   [8]  21 FF FF FF FF FF FF FF"],"fields":{"SID":139,"Instance":0,"DC Type":"Battery"},"description":"DC Detailed Status","timestamp":"2021-07-29T11:12:22.161Z"}

It would seem that when canboatjs does not get a full set of data, rather than ignoring it sends out a partial set of data.

It would seem the first of the two messages from device 227 (0x23) was lost and only the second part wads received with a fast packet reference of 0x81 and that the 0x80 was lost..

can1  19F212E3   [8]  81 FF FF FF FF FF FF FF

canboat seems to error, 

INFO 2021-07-29T11:21:47.652Z [analyzer] Assuming normal format with one line per packet
INFO 2021-07-29T11:21:47.652Z [analyzer] New PGN 127506 for device 227 (heap 11154 bytes)
ERROR 2021-07-29T11:21:47.652Z [analyzer] PGN 127506 malformed packet from 227 received; expected 1 but got 129

while canboatjs sent a partial set off data.

I believe the reason we get the *timeRemaining element in SignalK is due to the fact that the the PGN definition within SignalK has timeRemaining defined with allowNull being true.

If further details or debugging is required then please let me know.

I have also attached a recent SignalK screen shot where you can see a load of incorrect entries

[image]

nmbath, 2021-07-29:
Further to the above,

these four messages

can1 19F21229   [8]  40 0B F1 00 00 63 FF 40
can1 19F21229   [8]  41 38 FF FF 90 01 FF FF
can1 19F21229   [8]  61 38 FF FF 90 01 FF FF
can1 19F21229   [8]  80 0B 31 00 00 63 FF FF

created this output.

{"canId":435294761,"prio":6,"src":41,"dst":255,"pgn":127506,"timestamp":"2021-07-29T12:54:31.588Z","input":[],"fields":{"SID":241,"Instance":0,"DC Type":"Battery","State of Charge":99,"Time Remaining":14400},"description":"DC Detailed Status"}
{"canId":435294761,"prio":6,"src":41,"dst":255,"pgn":127506,"timestamp":"2021-07-29T12:54:31.588Z","fields":{"SID":97,"Instance":56,"State of Health":144,"Time Remaining":65281},"description":"DC Detailed Status"}

The second instance of 127506 with Instance 65 seems to correlate to the third message (61 38 FF FF 90 01 FF FF). Almost seems like at times the code is treating the packet as a single packet and using the first two bytes to reference the SID and instance.

Could there be a correlation with https://github.com/canboat/canboatjs/issues/161

nmbath, 2021-08-30:
Having dug further and gained information from the SignalK-dev slack discussions the following is a summary of PGNS impacted.

127497		Fast		9 bytes		
127506		Fast		11 bytes		
127507		Fast		6 bytes		
127508		Single	8 bytes		More investigation
127750		Single	8 bytes		More investigation

further investigation on PGN127508 and PGN127750 as it cant be associated with lost CAN messages as they are a single packet. I am certainly getting sporadic incorrect instance arriving, similar to the lost 2nd packet of a fast packet.  See screen shot, as the current and voltage from instance 96 is PGN127508.

[image]

sbender9, 2021-09-18:
Canboat and canboatjs act the pretty much the same. Both end up outputting the empty data for instance 0. 

I'm really not sure if this is because of an issue in canboat or an issue with the data being send incorrectly.

@keesverruijt any insight?

sbender9, 2021-09-18:
ok, this is clearly bad data coming from the device:

(2021-07-29 11:00:38.119447) can1 RX - - 19F21226 [8] 20 0B **8B 00 00** FF FF FF
(2021-07-29 11:00:38.120107) can1 RX - - 19F21226 [8] 21 FF FF FF FF FF FF FF
(2021-07-29 11:00:38.121154) can1 RX - - 19F21226 [8] 20 0B 8B 01 03 FF FF FF

That's SID 130, Instance 0, Battery Type 0

Not sure why it's sending that??

sbender9, 2021-09-18:
> Further to the above,
>
> these four messages
>
> can1 19F21229 [8] 40 0B F1 00 00 63 FF 40
> can1 19F21229 [8] 41 38 FF FF 90 01 FF FF
> can1 19F21229 [8] 61 38 FF FF 90 01 FF FF
> can1 19F21229 [8] 80 0B 31 [...]

Yes, the fix for #161 takes care of this.

dbeckwith01, 2021-09-24:
@sbender9 Is there a fix for #161 yet?

nmbath, 2021-09-28:
@sbender9, do you have any fix yet that can be tested?

sbender9, 2022-02-24:
this is fixed, right?

nmbath, 2022-02-24:
Yes, I believe we have. It can be closed and if further issues arise a new issue created or this one re-opened.

sbender9, 2022-02-24:
Fix published today in 1.22.0

closed 2022-02-24 by sbender9