canboat/canboatjs #172: PNG 128259, “Speed” value mismatch

cdyellick, 2021-12-31:
For PNG 128259, “Speed” I’m noticing a mismatch between the value returned by canboatjs’ parseString() and the value displayed by my instrument displays.  Given that this is a relatively primitive PNG, I’m assuming that this may be user error.

Take for example, the below message string that came from an Airmar DST810 via a Yacht Devices YDWG-02.  

`00:13:24.317 R 9F50323 FF 17 00 FF FF 00 FF FF`

Using the below code,

```
const FromPgn = require("@canboat/canboatjs").FromPgn;                                                                                                          
                                                                                                                                                                
const parser = new FromPgn();                                                                                                                                   
                                                                                                                                                                
const json = parser.parseString(                                                                                                                                
   "00:13:24.317 R 9F50323 FF 17 00 FF FF 00 FF FF"                                                                                                             
);                 
```

canboatjs returns the following:

```
{
    "canId": 167052067,
    "prio": 2,
    "src": 35,
    "dst": 255,
    "pgn": 128259,
    "direction": "R",
    "time": "00:13:24.317",
    "input": [
        "00:13:24.317 R 9F50323 FF 17 00 FF FF 00 FF FF"
    ],
    "fields": {
        "Speed Water Referenced": 0.23,
        "Speed Water Referenced Type": "Paddle wheel"
    },
    "description": "Speed",
    "timestamp": "2021-12-31T05:04:31.271Z"
}
```

Notably it shows that the Speed Water Referenced is 0.23, which is simply the decimal representation of the second byte (17) divided by 100.

However, when that same message is sent to my test NMEA 2000 network, made up of a YDWG-02 and a B&G Triton display, the Triton display shows 0.4 knots.

Here is the code (Python) that I used to put the message onto the NMEA 2000 network, using the YDWG-02.

```
import time

import serial

SERIAL_PORT = '/dev/cu.usbmodem004012461'
ser = serial.Serial(SERIAL_PORT)

for x in range(1, 100):
    ser.write('9F50323 FF 17 00 FF FF 00 FF FF\r\n'.encode())
    time.sleep(1)
```

 Does anyone know why I’m getting these two seemingly incongruous results?

sbender9, 2021-12-31:
It's in m/s in the N2K message.

cdyellick, 2021-12-31:
Ah, okay that makes sense.  Now that I'm taking a second look at the corresponding entry in pngs.json, I see that it specifies the units as m/s, so I should have been able to figure that out.

Thank you for the quick response to what ended up being a dumb question.

sbender9, 2021-12-31:
Jcanboatjs will alway use SI units. So even if pgns.json had knots, it would use m/s.

closed 2021-12-31 by sbender9

cdyellick, 2021-12-31:
That's good to know; thank you.