canboat/canboatjs #192: PGN 127489

fencer, 2022-04-17:
I was analysing the incoming data from Mercury VesselView link and noticed that the PGN 127489 was not being decoded (using analyser). Here is some samples

```
1556284840920;A;2019-04-26T13:20:40.920Z,2,127489,8,255,8,00,1a,00,ff,ff,ff,ff,8b
1556284840921;A;2019-04-26T13:20:40.921Z,2,127489,8,255,8,01,79,a5,04,0c,00,e4,a5
1556284840922;A;2019-04-26T13:20:40.921Z,2,127489,8,255,8,02,0c,00,fb,00,ff,ff,7f
```

sbender9, 2022-04-17:
that PGN is definitely supported. : https://github.com/canboat/canboat/blob/bf2b56cd1763e9a6258b60ea5752a0eacd890c9a/analyzer/pgn.h#L2876

sbender9, 2022-04-17:
2019-04-26T13:20:40.920Z,2,127489,8,255,8,00,**1a**,00,ff,ff,ff,ff,8b

0x1a means there should be 26 bytes, but there are only 20  bytes in your example data.

fencer, 2022-04-18:
> that PGN is definitely supported. : https://github.com/canboat/canboat/blob/bf2b56cd1763e9a6258b60ea5752a0eacd890c9a/analyzer/pgn.h#L2876

Yes in fact I tried out the sentences in the tests and can decode them using analyser but cannot not any of the samples I took from the log 
[image]

sbender9, 2022-04-22:
Either your device is not forming the PGNs correctly, or you're not getting all the frames in your example data above.

1556284840920;A;2019-04-26T13:20:40.920Z,2,127489,8,255,8,**00**,1a,00,ff,ff,ff,ff,8b
1556284840921;A;2019-04-26T13:20:40.921Z,2,127489,8,255,8,**01**,79,a5,04,0c,00,e4,a5
1556284840922;A;2019-04-26T13:20:40.921Z,2,127489,8,255,8,**02**,0c,00,fb,00,ff,ff,7f`

You sure there's not another 4th frame in your data?

sbender9, 2022-04-22:
You also need to send all the frames to analyzer at the same time. You can't send them separately.

Note that those are not three different instances of the pgn sent. There are multiple lines for each message.

closed 2022-06-09 by sbender9