canboat/canboatjs #195: installing on raspi

leeprevost, 2022-07-10:
I'm building an embedded applicaiton running on a raspberry pi zero w running latest "Debian GNU/Linux 11 (bullseye)."  

I'm having trouble getting canboat installed on a fresh pi.   My steps:

- updated debian using update&&upgrade
- installed npm
- updated npm including npm@latest -g
- sudo npm intall @canboat/canboatjs

Gettting:
```
pi@rpiz2w-lee:~ $ sudo npm install @canboat/canboatjs

up to date, audited 134 packages in 6s

21 packages are looking for funding
  run `npm fund` for details

5 high severity vulnerabilities

To address all issues (including breaking changes), run:
  npm audit fix --force

Run `npm audit` for details.
pi@rpiz2w-lee:~ $ npm audit fix

up to date, audited 134 packages in 7s

21 packages are looking for funding
  run `npm fund` for details

# npm audit report

glob-parent  <5.1.2
Severity: high
Regular expression denial of service in glob-parent - https://github.com/advisories/GHSA-ww39-953v-wcq6
fix available via `npm audit fix --force`
Will install @canboat/canboatjs@1.5.0, which is a breaking change
node_modules/glob-parent
  glob-stream  5.3.0 - 6.1.0
  Depends on vulnerable versions of glob-parent
  node_modules/glob-stream
    help-me  1.0.0 - 1.1.0
    Depends on vulnerable versions of glob-stream
    node_modules/help-me
      mqtt  1.14.1 - 4.2.6
      Depends on vulnerable versions of help-me
      node_modules/mqtt
        @canboat/canboatjs  >=1.5.1
        Depends on vulnerable versions of mqtt
        node_modules/@canboat/canboatjs

5 high severity vulnerabilities

To address all issues (including breaking changes), run:
  npm audit fix --force
pi@rpiz2w-lee:~ $ sudo npm audit fix

up to date, audited 134 packages in 5s

21 packages are looking for funding
  run `npm fund` for details

# npm audit report

glob-parent  <5.1.2
Severity: high
Regular expression denial of service in glob-parent - https://github.com/advisories/GHSA-ww39-953v-wcq6
fix available via `npm audit fix --force`
Will install @canboat/canboatjs@1.5.0, which is a breaking change
node_modules/glob-parent
  glob-stream  5.3.0 - 6.1.0
  Depends on vulnerable versions of glob-parent
  node_modules/glob-stream
    help-me  1.0.0 - 1.1.0
    Depends on vulnerable versions of glob-stream
    node_modules/help-me
      mqtt  1.14.1 - 4.2.6
      Depends on vulnerable versions of help-me
      node_modules/mqtt
        @canboat/canboatjs  >=1.5.1
        Depends on vulnerable versions of mqtt
        node_modules/@canboat/canboatjs

5 high severity vulnerabilities

To address all issues (including breaking changes), run:
  npm audit fix --force
pi@rpiz2w-lee:~ $ sudo npm audit fix --force
npm WARN using --force Recommended protections disabled.
npm WARN audit Updating @canboat/canboatjs to 1.5.0, which is a SemVer major change.

removed 73 packages, changed 23 packages, and audited 16 packages in 43s

# npm audit report

set-value  3.0.0 - 4.0.0
Severity: high
Prototype Pollution in set-value - https://github.com/advisories/GHSA-4jqc-8m5r-9rpr
fix available via `npm audit fix`

  prompt-choices  >=4.1.0
  Depends on vulnerable versions of set-value

2 high severity vulnerabilities

To address all issues, run:
  npm audit fix
```

I'm a python guy and don't know my node/npm cli well.  See anyting I'm doing wrong?

sbender9, 2022-07-10:
I don't see any errors, but not sure if you really want "npm install"?

"npm install -g" will get you the command line utilities in your path.

leeprevost, 2022-07-10:
OK -- could be my unfamiliarity with node/npm environment.   But, I'm following your instructions on readme: npm install @canboat/canboatjs

Also, maybe I can now run this but I assumed my inability to get past the vulnerabilities and audit meant it didn't install correctly?

leeprevost, 2022-07-10:
I just read docs on npm install.  I see what you are telling me now about -g flag to install globally.  I'll try that as I am trying to run it via command line using nc ydwg <port> | analyzejs

sbender9, 2022-07-10:
Yep. 

Those audit/warnings are normal.

leeprevost, 2022-07-10:
Thank you. I do seem to be running now.   Thank you for your help and patience.

closed 2022-07-10 by leeprevost