canboat/canboatjs #220: Support for 'standard' length CAN messages

grazzam, 2023-04-17:
I have a REC ABMS that outputs a standard CAN frame (11-bit identifier) instead of the "extended" CAN frame (29-bit identifier) required by NMEA 2000. Is it possible to parse these sentences? Besides finding out what the sentences are from the manufacturer, creating entries in pgns.json, I assume this would also affect the analyzer.

[image]

grazzam, 2023-04-18:
Attached image shows a few seconds of the output of "candump can0 | candump2analyzer | analyzer" with these 'standard' length CAN sentences. Seems like analyzer tries to assign new pgns and device IDs based on mis-interpreting the data.

[image]

sbender9, 2023-04-18:
You are using canboat here, not canboatjs. Have you tried with canboatjs?

Otherwise you should open an issue on the canboat git.

sbender9, 2023-04-18:
And BTW, don't think canboatjs could handle this either.

grazzam, 2023-04-18:
Ah thanks, agreed I should move this and that although it turns out I have
not tried it with canboatjs, it would likely have similar trouble.

closed 2023-04-18 by grazzam