canboat/canboatjs #242: TypeError: SerialPort is not a constructor

gabsi55, 2023-08-10:
With a recent SignalK installation via NodeJS i encountered the following error. What can I do against it?
Thanks in advance.

TypeError: SerialPort is not a constructor
    at SerialStream.start (/usr/local/lib/node_modules/signalk-server/node_modules/@canboat/canboatjs/lib/serial.js:116:19)
    at new SerialStream (/usr/local/lib/node_modules/signalk-server/node_modules/@canboat/canboatjs/lib/serial.js:90:8)
    at Object.nmea2000input [as NMEA2000] (/usr/local/lib/node_modules/signalk-server/node_modules/@signalk/streams/simple.js:242:7)
    at new Simple (/usr/local/lib/node_modules/signalk-server/node_modules/@signalk/streams/simple.js:82:34)
    at createPipeElement (/usr/local/lib/node_modules/signalk-server/lib/pipedproviders.js:75:16)
    at /usr/local/lib/node_modules/signalk-server/lib/pipedproviders.js:48:30
    at Array.reduce (<anonymous>)
    at createPipedProvider (/usr/local/lib/node_modules/signalk-server/lib/pipedproviders.js:46:55)
    at /usr/local/lib/node_modules/signalk-server/lib/pipedproviders.js:82:37
    at Array.reduce (<anonymous>)
    at Object.startProviders [as start] (/usr/local/lib/node_modules/signalk-server/lib/pipedproviders.js:79:62)
    at Server.<anonymous> (/usr/local/lib/node_modules/signalk-server/lib/index.js:329:66)
    at Generator.next (<anonymous>)
    at fulfilled (/usr/local/lib/node_modules/signalk-server/lib/index.js:23:58)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)

sbender9, 2023-08-10:
What version of nodejs are you running?

I think something is off with your install. Try removing the server and installing again.

https://github.com/SignalK/signalk-server/wiki/FAQ:-Frequently-Asked-Questions#somethings-wrong-i-need-to-reinstall--reset-my-system

gabsi55, 2023-08-10:
I have Node.js v18.17.1 on an ubuntu system.

I also tried a reinstall but did not solve the problem.
Is this setup with nodejs running in ubuntu environments?

sbender9, 2023-08-10:
Normally should be fine on ubuntu, but not sure about serialport support for your hardware.

Can you check `/usr/local/lib/node_modules/signalk-server/node_modules/serialport/package.json` to see what version is installed?

Also check `/usr/local/lib/node_modules/signalk-server/node_modules/@canboat/canboatjs/node_modules/serialport/package.json`

gabsi55, 2023-08-10:
It is a Actisense NGT-1.. With a serial terminal a can open a connection.
Serialport version: 11.0.1

The second directory is not existing.

Thanks for your help.

tkurki, 2023-08-10:
Scott, I think this is what has happened here (and elsewhere):

Previously server and canboatjs were using serialport@9. Npm was smart enough to detect that, depuplicated serialport and it was only installed under signalk-server/node_modules and canboatjs at signalk-server/node_modules/@canboat/canboatjs would pick it from there.

When server switched to serialport@11 npm was not smart enough *when updating the server* to detect that canboatjs's dependency is no longer met properly and here we are, canboatjs invoking serialport@9 api but loading serialport@11.

This is why you can't replicate the situation just by installing signalk-server. Should be easy enough to verify, but commuting right now....

I think running `npm install -g signalk-server` should fix things.

sbender9, 2023-08-10:
oddly enough, I get the same error with canboatjs 2.1 . But you should not have that.

Can you check the canboatjs version?

sbender9, 2023-08-10:
ok., they released a new serialport 11.0.1 a couple weeks ago. I think this might be related. trying to figure this out...

gabsi55, 2023-08-10:
It worked with `sudo npm i -g @canboat/canboatjs@1.28.2` i can install it. But actisense-serialjs fails with 

Error: Cannot find module 'serialport'

gabsi55, 2023-08-10:
By installing serialport V10.0.0 i got it to work.

sbender9, 2023-08-10:
I just published canboatjs v1.28.3 that should fix this

just do `sudo npm install -g signalk-server` and you should get the new version

gabsi55, 2023-08-11:
Working perfect. Thanks

closed 2023-08-11 by gabsi55