canboat/canboatjs #246: Hi there, a Newby here. I need some help

zuchi238, 2023-09-15:
My overall plan is to ask my boat "Are we on course?" and then let the boat reply (over speaker.
My best buddy in this is ... chatGPT ;) ( I have no formal training in coding what so ever)
So after a week I came this far:
1. I'm able to read date via my Yacht Devices YDNU 02 (via serial port)
2. I was able to install canboatjs
3. I was able to update canboatjs to decode the NMEA sentences.

(via this code:

const SerialPort = require('serialport').SerialPort;
const FromPgn = require('@canboat/canboatjs').FromPgn;

const port = new SerialPort({ path: '/dev/tty.usbmodem004017681', baudRate: 9600 });

const parser = new FromPgn();

port.on('open', () => {
  console.log(`Serial port /dev/tty.usbmodem004017681 is open.`);
});

port.on('data', (data) => {
  const dataString = data.toString();
  console.log('Received data:', dataString);

  const json = parser.parseString(dataString);
  if (json) {
    console.log('Decoded NMEA data:', JSON.stringify(json));
  }
});

port.on('error', (err) => {
  console.log('Error:', err.message);
});

Then when I ran the code I got this message:

(base) bartstevens@barts-air v5 % node read_nmea_data2.js
Serial port /dev/tty.usbmodem004017681 is open.
Received data: $YDROT,0.5,A*3E
$YDHDG,257.9,,,,*5D
$YDHDM,257.9,M*36
$YDRSA,-22.2,A,,V*7B

Error parsing data: Error [ERR_UNHANDLED_ERROR]: Unhandled error. ('$YDROT,0.5,A*3E\r\n' +
  '$YDHDG,257.9,,,,*5D\r\n' +
  '$YDHDM,257.9,M*36\r\n' +
  '$YDRSA,-22.2,A,,V*7B\r\n')
Received data: $YDGSV,3,1,12,10,23,299,00,12,43,217,00,13,22,145,00,14,06,052,00*76
$YDGSV,3,2,12,15,44,182,00,17,32,061,00,19,37,093,00,22,25,053,00*79
$YDGSV,3,3,12,23,23,259,00,24,80,294,00,25,04,229,00,74,18,253,00*7B
$YDGSA,A,1,,,,,,,,,,,,,,,,1*09
$YDXDR,A,-1.78,D,Pitch,A,1.87,D,Roll*05
$YDHDG,257.9,,,,*5D

Error parsing data: Error [ERR_UNHANDLED_ERROR]: Unhandled error. ('$YDGSV,3,1,12,10,23,299,00,12,43,217,00,13,22,145,00,14,06,052,00*76\r\n' +
  '$YDGSV,3,2,12,15,44,182,00,17,32,061,00,19,37,093,00,22,25,053,00*79\r\n' +
  '$YDGSV,3,3,12,23,23,259,00,24,80,294,00,25,04,229,00,74,18,253,00*7B\r\n' +
  '$YDGSA,A,1,,,,,,,,,,,,,,,,1*09\r\n' +
  '$YDXDR,A,-1.78,D,Pitch,A,1.87,D,Roll*05\r\n' +
  '$YDHDG,257.9,,,,*5D\r\n')
Received data: $YDROT,-1.0,A*17
$YDHDG,257.9,,,2.2,E*36
$YDHDM,257.9,M*36
$YDHDT,260.1,T*3A
$YDRSA,-22.2,A,,V*7B

Error parsing data: Error [ERR_UNHANDLED_ERROR]: Unhandled error. ('$YDROT,-1.0,A*17\r\n' +
  '$YDHDG,257.9,,,2.2,E*36\r\n' +
  '$YDHDM,257.9,M*36\r\n' +
  '$YDHDT,260.1,T*3A\r\n' +
  '$YDRSA,-22.2,A,,V*7B\r\n')
Received data: $YDGSV,3,1,12,10,23,299,00,12,43,217,00,13,22,145,00,14,06,052,00*76
$YDGSV,3,2,12,15,44,182,00,17,32,061,00,19,37,093,00,22,25,053,00*79
$YDGSV,3,3,12,23,23,259,00,24,80,294,00,25,04,229,00,74,18,253,00*7B
$YDGSA,A,1,,,,,,,,,,,,,,,,1*09
$YDXDR,A,-1.78,D,Pitch,A,1.87,D,Roll*05
$YDHDG,257.9,,,2.2,E*36

ChatGPT tells me this : 

The "Unhandled error" messages suggest that the parser.parseString() function is expecting a different format of data than what it's receiving. The data from your device seems to be standard NMEA 0183 sentences, not NMEA 2000 PGNs that the canboatjs parser is designed to handle.

Since you're reading NMEA 0183 data, you don't need the canboatjs library to parse it. NMEA 0183 sentences are simple text-based sentences that can be parsed without any specialized libraries, though there are libraries designed to make this easier if you prefer.

My Hardware consists of:

B&G material (Zeus3, VHF, Rudder sensor, Precision 9 compass, and several Yacht Devices (YDNU, MOB button, YDSC 04, YDCC 04 and YDNR)

So ... who can go beyond ChatGPT and explain what I'm doing wrong so far.

Once I have this figured out I want to create a bridge to Python and link a voice library via Python so I'm able to ask "e.g. Am I on course?"

Hope somebody can point me (and ChatGPT) in the right correction.

Be well,

Bart

tkurki, 2023-09-15:
ChatGPT gave you exactly the correct answer: Canboatjs is for NMEA2000 data and your data is NMEA 0183.

I already suggest elsewhere to check out [Signal K server](https://github.com/SignalK/signalk-server). Install it, configure a NMEA0183 data connection with serial input. Then the Data Browser[(demo here)](https://demo.signalk.io/admin/#/databrowser) will show what data you have in an understandable format. Then you can use the Signal K http API [(demo here)](https://demo.signalk.io/signalk/v1/api/vessels/self/navigation) to access the data from Python.

closed 2023-09-15 by tkurki

zuchi238, 2023-09-15:
Dear Tkurki,
Good morning! Thanks for taking the time to answer my question.

Some remarks and follow up question to you (as an expert)

1. I thought wanted to keep it simple for now ... Just a python script bridged to CanboatJS ...
What you say is to use SignalK to collect the data via the USB, and then use a Python script to ask (my famous) question (via voice? So run signalK just in the background?
2. The strange thing is that I have a NMEA bus, with new B&G material on it. But I still (only get NMEA 0183 data ... How come??
Let me know your thoughts.

Best 

Bart

zuchi238, 2023-09-15:
(and when I get NMEA 0183 a d NMEA 2000 data simultaneously ? Is that possible?)

zuchi238, 2023-09-15:
Dear Tkurki,

What a cool project, I have signalK installed, linked it to my YDNU 02, get data in ... and it works almost "Out of the box" (with my good friend ChatGPT)
Now I'm creating a python script to get some data out of SK. This seams to work as well with a Websocket.
Next will be to connect voice into the Python script so I can talk ... will keep you posted and always open to ideas!
Bart

tkurki, 2023-09-15:
Read the docs for your YDNU for instructions re: using NMEA2000. Be sure to recreate the connection in sk if you do switch.

zuchi238, 2023-09-15:
What do you mean with " ...Be sure to recreate the connection in sk if you do switch ..." ?

tkurki, 2023-09-15:
If you switch to using nmea2000 the nmea0183 connection will stop working.

zuchi238, 2023-09-15:
Since I have all my instruments on a NMEA 2000 bus, connected to a B&G Zeus 3, isn't all data on the bus then not all 2000?
So not a bit 0183 and a bit 2000?
Bart

tkurki, 2023-09-16:
>Read the docs for your YDNU