canboat/canboatjs #263: Possible bug in canbus.js?

miggat, 2024-03-07:
Hello, 
I'm not sure if this is the correct place to post the issue, but i'm facing this error when starting SignalK:

Mar 07 13:02:16 Trace at new CanbusStream (/usr/lib/node_modules/signalk-server/node_modules/@canboat/canboatjs/lib/canbus.js:57:12) at CanbusStream (/usr/lib/node_modules/signalk-server/node_modules/@canboat/canboatjs/lib/canbus.js:46:12) at Object.nmea2000input [as NMEA2000] (/usr/lib/node_modules/signalk-server/node_modules/@signalk/streams/simple.js:259:30) at new Simple (/usr/lib/node_modules/signalk-server/node_modules/@signalk/streams/simple.js:92:34) at createPipeElement (/usr/lib/node_modules/signalk-server/lib/pipedproviders.js:103:16) at /usr/lib/node_modules/signalk-server/lib/pipedproviders.js:72:30 at Array.reduce (<anonymous>) at createPipedProvider (/usr/lib/node_modules/signalk-server/lib/pipedproviders.js:70:55) at /usr/lib/node_modules/signalk-server/lib/pipedproviders.js:110:37 at Array.reduce (<anonymous>) at Object.startProviders [as start] (/usr/lib/node_modules/signalk-server/lib/pipedproviders.js:107:62) at Server.<anonymous> (/usr/lib/node_modules/signalk-server/lib/index.js:321:75) at Generator.next (<anonymous>) at fulfilled (/usr/lib/node_modules/signalk-server/lib/index.js:23:58) at process.processTicksAndRejections (node:internal/process/task_queues:95:5)

After this error, there's no N2K input or output.

Also, I'm not sure if this is a problem with this library, the plugin itself, or signalk

miggat, 2024-03-07:
In the dashboard, the connection appears completely idle. No deltas at all.

I'm not sure if it's worth mentioning, but this are the firsts lines after a server restart:
```

Mar 07 13:18:37 [signalk-to-nmea0183] GGA: no position, not converting
Mar 07 13:18:40 alarmsilencer:no configuration data
Mar 07 13:18:45 Trace at new CanbusStream (/usr/lib/node_modules/signalk-server/node_modules/@canboat/canboatjs/lib/canbus.js:57:12) at CanbusStream (/usr/lib/node_modules/signalk-server/node_modules/@canboat/canboatjs/lib/canbus.js:46:12) at Object.nmea2000input [as NMEA2000] (/usr/lib/node_modules/signalk-server/node_modules/@signalk/streams/simple.js:259:30) at new Simple (/usr/lib/node_modules/signalk-server/node_modules/@signalk/streams/simple.js:92:34) at createPipeElement (/usr/lib/node_modules/signalk-server/lib/pipedproviders.js:103:16) at /usr/lib/node_modules/signalk-server/lib/pipedproviders.js:72:30 at Array.reduce (<anonymous>) at createPipedProvider (/usr/lib/node_modules/signalk-server/lib/pipedproviders.js:70:55) at /usr/lib/node_modules/signalk-server/lib/pipedproviders.js:110:37 at Array.reduce (<anonymous>) at Object.startProviders [as start] (/usr/lib/node_modules/signalk-server/lib/pipedproviders.js:107:62) at Server.<anonymous> (/usr/lib/node_modules/signalk-server/lib/index.js:321:75) at Generator.next (<anonymous>) at fulfilled (/usr/lib/node_modules/signalk-server/lib/index.js:23:58) at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
Mar 07 13:18:45 undefined
Mar 07 13:18:46 signalk-server running at 0.0.0.0:[object Object]
Mar 07 13:18:46 /home/pi/.signalk/resources - OK...
Mar 07 13:18:46 ** FS createSavePaths() **
Mar 07 13:18:46 /home/pi/.signalk/resources/waypoints - OK....
Mar 07 13:18:46 /home/pi/.signalk/resources/routes - OK....
Mar 07 13:18:46 /home/pi/.signalk/resources/notes - OK....
Mar 07 13:18:46 /home/pi/.signalk/resources/regions - OK....
Mar 07 13:18:47 Handle: {"path":"navigation.position","value":{"longitude":1.7284883333333334,"latitude":41.212865}}
Mar 07 13:18:47 Handle: {"path":"environment.wind.directionTrue","value":0.3836233696759462}
Mar 07 13:18:47 AggregateError at internalConnectMultiple (node:net:1114:18) at afterConnectMultiple (node:net:1667:5) { code: 'ETIMEDOUT', [errors]: [ Error: connect ETIMEDOUT 127.0.0.1:8086 at createConnectionError (node:net:1634:14) at Timeout.internalConnectMultipleTimeout (node:net:1685:38) at listOnTimeout (node:internal/timers:575:11) at process.processTimers (node:internal/timers:514:7) { errno: -110, code: 'ETIMEDOUT', syscall: 'connect', address: '127.0.0.1', port: 8086 }, Error: connect ECONNREFUSED ::1:8086 at createConnectionError (node:net:1634:14) at afterConnectMultiple (node:net:1664:40) { errno: -111, code: 'ECONNREFUSED', syscall: 'connect', address: '::1', port: 8086 } ] }
Mar 07 13:18:48 [object Object]
```

miggat, 2024-03-08:
I noticed all this in the logs, but the n2k devices dont receive anything
```
Mar 08 12:46:01 2024-03-08T11:46:01.392Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.611,5.556001407520357
Mar 08 12:46:01 2024-03-08T11:46:01.394Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.556001407520357,"Wind Angle":5.6721853071795865,"Reference":2}
Mar 08 12:46:02 2024-03-08T11:46:02.321Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.585,5.556001407520357
Mar 08 12:46:02 2024-03-08T11:46:02.322Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.556001407520357,"Wind Angle":5.698185307179586,"Reference":2}
Mar 08 12:46:02 2024-03-08T11:46:02.359Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.585,5.453112492566276
Mar 08 12:46:02 2024-03-08T11:46:02.360Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.453112492566276,"Wind Angle":5.698185307179586,"Reference":2}
Mar 08 12:46:03 2024-03-08T11:46:03.288Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.593,5.453112492566276
Mar 08 12:46:03 2024-03-08T11:46:03.289Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.453112492566276,"Wind Angle":5.690185307179586,"Reference":2}
Mar 08 12:46:03 2024-03-08T11:46:03.309Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.593,5.2473346626581145
Mar 08 12:46:03 2024-03-08T11:46:03.310Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.2473346626581145,"Wind Angle":5.690185307179586,"Reference":2}
Mar 08 12:46:04 2024-03-08T11:46:04.260Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.576,5.2473346626581145
Mar 08 12:46:04 2024-03-08T11:46:04.261Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.2473346626581145,"Wind Angle":5.707185307179587,"Reference":2}
Mar 08 12:46:04 2024-03-08T11:46:04.285Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.576,5.2473346626581145
Mar 08 12:46:04 2024-03-08T11:46:04.286Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.2473346626581145,"Wind Angle":5.707185307179587,"Reference":2}
Mar 08 12:46:05 2024-03-08T11:46:05.326Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.646,5.2473346626581145
Mar 08 12:46:05 2024-03-08T11:46:05.328Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.2473346626581145,"Wind Angle":5.637185307179586,"Reference":2}
Mar 08 12:46:05 2024-03-08T11:46:05.347Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.646,5.556001407520357
Mar 08 12:46:05 2024-03-08T11:46:05.349Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.556001407520357,"Wind Angle":5.637185307179586,"Reference":2}
Mar 08 12:46:06 2024-03-08T11:46:06.195Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.672,5.556001407520357
Mar 08 12:46:06 2024-03-08T11:46:06.197Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.556001407520357,"Wind Angle":5.6111853071795865,"Reference":2}
Mar 08 12:46:06 2024-03-08T11:46:06.219Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.672,5.6074458649973975
Mar 08 12:46:06 2024-03-08T11:46:06.220Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.6074458649973975,"Wind Angle":5.6111853071795865,"Reference":2}
Mar 08 12:46:06 GET /signalk/v1/api/vessels/self/uuid 404 5.532 ms - 171
Mar 08 12:46:07 2024-03-08T11:46:07.181Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.707,5.6074458649973975
Mar 08 12:46:07 2024-03-08T11:46:07.181Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.6074458649973975,"Wind Angle":5.576185307179586,"Reference":2}
Mar 08 12:46:07 2024-03-08T11:46:07.198Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.707,5.453112492566276
Mar 08 12:46:07 2024-03-08T11:46:07.199Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.453112492566276,"Wind Angle":5.576185307179586,"Reference":2}
Mar 08 12:46:08 2024-03-08T11:46:08.133Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.663,5.453112492566276
Mar 08 12:46:08 2024-03-08T11:46:08.134Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.453112492566276,"Wind Angle":5.620185307179586,"Reference":2}
Mar 08 12:46:08 2024-03-08T11:46:08.151Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.663,5.504556950043316
Mar 08 12:46:08 2024-03-08T11:46:08.152Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.504556950043316,"Wind Angle":5.620185307179586,"Reference":2}
Mar 08 12:46:09 2024-03-08T11:46:09.154Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.698,5.504556950043316
Mar 08 12:46:09 2024-03-08T11:46:09.157Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.504556950043316,"Wind Angle":5.585185307179586,"Reference":2}
Mar 08 12:46:09 2024-03-08T11:46:09.176Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.698,5.504556950043316
[... cut ...]
Mar 08 12:46:20 2024-03-08T11:46:20.732Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.813223694905559,"Wind Angle":5.6111853071795865,"Reference":2}
Mar 08 12:46:20 2024-03-08T11:46:20.761Z signalk-to-nmea2000 environment.wind.angleApparent,environment.wind.speedApparent:-0.672,5.864668152382599
Mar 08 12:46:20 2024-03-08T11:46:20.762Z signalk-to-nmea2000 emit nmea2000JsonOut {"pgn":130306,"Wind Speed":5.864668152382599,"Wind Angle":5.6111853071795865,"Reference":2}
Mar 08 12:46:20 GET /signalk/v1/api/atons/ 200 4.628 ms - 2467
```

sbender9, 2024-03-08:
Can you please trying the following:

`sudo npm install -g @canboat/canboatjs`

`candumpjs can0`

Do you get anything? 

If you want to take canboatjs out of the picture, try `candump can0` . (You make need to install cantools: `sudo apt install can-utils`

miggat, 2024-03-08:
I will try tomorrow and report back. 
If is of any help, I already tried the candump that comes with openplotter and it shows there's traffic, but reinstalling canboatjs seems like a good option I didn't thought about.

sbender9, 2024-03-08:
That install won't effect Signal K, but trying candumpjs may give us some insight into the issue...

miggat, 2024-03-10:
I tried installing and running candumpjs and this is the result, which seems totally fine :S

```
pi@openplotter:~ $ sudo npm install -g @canboat/canboatjs
npm ERR! code EEXIST
npm ERR! path /usr/bin/ikonvert-serial
npm ERR! EEXIST: file already exists
npm ERR! File exists: /usr/bin/ikonvert-serial
npm ERR! Remove the existing file and try again, or run npm
npm ERR! with --force to overwrite files recklessly.

npm ERR! A complete log of this run can be found in: /root/.npm/_logs/2024-03-10T10_25_41_347Z-debug-0.log
pi@openplotter:~ $ sudo npm install -g --force @canboat/canboatjs
npm WARN using --force Recommended protections disabled.

added 209 packages in 1m

31 packages are looking for funding
  run `npm fund` for details
pi@openplotter:~ $ candumpjs can0
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:29:56.872Z","input":["2024-03-10T10:29:56.872Z,3,130306,100,255,8,ff,68,01,38,09,fa,ff,ff"],"fields":{"Wind Speed":3.6,"Wind Angle":0.236,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:29:56.901Z","input":["2024-03-10T10:29:56.901Z,3,130306,100,255,8,ff,68,01,38,09,fa,ff,ff"],"fields":{"Wind Speed":3.6,"Wind Angle":0.236,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:29:57.818Z","input":["2024-03-10T10:29:57.818Z,3,130306,100,255,8,ff,68,01,da,07,fa,ff,ff"],"fields":{"Wind Speed":3.6,"Wind Angle":0.201,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:29:57.837Z","input":["2024-03-10T10:29:57.837Z,3,130306,100,255,8,ff,7d,01,da,07,fa,ff,ff"],"fields":{"Wind Speed":3.81,"Wind Angle":0.201,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:29:58.791Z","input":["2024-03-10T10:29:58.791Z,3,130306,100,255,8,ff,7d,01,66,12,fa,ff,ff"],"fields":{"Wind Speed":3.81,"Wind Angle":0.471,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:29:58.837Z","input":["2024-03-10T10:29:58.837Z,3,130306,100,255,8,ff,b0,01,66,12,fa,ff,ff"],"fields":{"Wind Speed":4.32,"Wind Angle":0.471,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:29:59.758Z","input":["2024-03-10T10:29:59.758Z,3,130306,100,255,8,ff,b0,01,a6,0e,fa,ff,ff"],"fields":{"Wind Speed":4.32,"Wind Angle":0.375,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:29:59.775Z","input":["2024-03-10T10:29:59.775Z,3,130306,100,255,8,ff,c5,01,a6,0e,fa,ff,ff"],"fields":{"Wind Speed":4.53,"Wind Angle":0.375,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:30:00.722Z","input":["2024-03-10T10:30:00.722Z,3,130306,100,255,8,ff,c5,01,5a,0f,fa,ff,ff"],"fields":{"Wind Speed":4.53,"Wind Angle":0.393,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:30:00.755Z","input":["2024-03-10T10:30:00.755Z,3,130306,100,255,8,ff,c0,01,5a,0f,fa,ff,ff"],"fields":{"Wind Speed":4.48,"Wind Angle":0.393,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:30:01.691Z","input":["2024-03-10T10:30:01.691Z,3,130306,100,255,8,ff,c0,01,fc,0d,fa,ff,ff"],"fields":{"Wind Speed":4.48,"Wind Angle":0.358,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:30:01.742Z","input":["2024-03-10T10:30:01.742Z,3,130306,100,255,8,ff,cf,01,fc,0d,fa,ff,ff"],"fields":{"Wind Speed":4.63,"Wind Angle":0.358,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:30:02.666Z","input":["2024-03-10T10:30:02.666Z,3,130306,100,255,8,ff,cf,01,bc,11,fa,ff,ff"],"fields":{"Wind Speed":4.63,"Wind Angle":0.454,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:30:02.693Z","input":["2024-03-10T10:30:02.693Z,3,130306,100,255,8,ff,b5,01,bc,11,fa,ff,ff"],"fields":{"Wind Speed":4.37,"Wind Angle":0.454,"Reference":"Apparent"},"description":"Wind Data"}
{"canId":234685028,"prio":3,"src":100,"dst":255,"pgn":130306,"timestamp":"2024-03-10T10:30:03.627Z","input":["2024-03-10T10:30:03.627Z,3,130306,100,255,8,ff,b5,01,d6,15,fa,ff,ff"],"fields":{"Wind Speed":4.37,"Wind Angle":0.559,"Reference":"Apparent"},"description":"Wind Data"}
```

candump also seems to work fine:

```
pi@openplotter:~ $ candump can0
  can0  0DFD0264   [8]  FF 82 01 F0 ED FA FF FF
  can0  0DFD0264   [8]  FF 72 01 F0 ED FA FF FF
  can0  0DFD0264   [8]  FF 72 01 12 F4 FA FF FF
  can0  0DFD0264   [8]  FF 68 01 12 F4 FA FF FF
  can0  0DFD0264   [8]  FF 68 01 B8 01 FA FF FF
  can0  0DFD0264   [8]  FF 59 01 B8 01 FA FF FF
  can0  0DFD0264   [8]  FF 59 01 AA 00 FA FF FF
  can0  0DFD0264   [8]  FF 63 01 AA 00 FA FF FF
  can0  0DFD0264   [8]  FF 63 01 62 02 FA FF FF
  can0  0DFD0264   [8]  FF 5E 01 62 02 FA FF FF
  can0  0DFD0264   [8]  FF 5E 01 68 F3 FA FF FF
  can0  0DFD0264   [8]  FF 59 01 68 F3 FA FF FF
```

This makes me even more lost :S

miggat, 2024-03-10:
I tried disabling ALL other SignalK plugins except for signalk-to-nmea200, and now I can see this error:

`Mar 10 11:57:18 Error: not found at /usr/lib/node_modules/signalk-server/lib/requestResponse.js:85:20 at new Promise (<anonymous>) at queryRequest (/usr/lib/node_modules/signalk-server/lib/requestResponse.js:82:12) at /usr/lib/node_modules/signalk-server/lib/serverroutes.js:271:44 at Layer.handle [as handle_request] (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/layer.js:95:5) at next (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/route.js:149:13) at Route.dispatch (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/route.js:119:3) at Layer.handle [as handle_request] (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/layer.js:95:5) at /usr/lib/node_modules/signalk-server/node_modules/express/lib/router/index.js:284:15 at param (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/index.js:365:14) at param (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/index.js:376:14) at Function.process_params (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/index.js:421:3) at next (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/index.js:280:10) at /usr/lib/node_modules/signalk-server/lib/tokensecurity.js:755:25 at Layer.handle [as handle_request] (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/layer.js:95:5) at trim_prefix (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/index.js:328:13) at /usr/lib/node_modules/signalk-server/node_modules/express/lib/router/index.js:286:9 at Function.process_params (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/index.js:346:12) at next (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/index.js:280:10) at cookieParser (/usr/lib/node_modules/signalk-server/node_modules/cookie-parser/index.js:57:14) at Layer.handle [as handle_request] (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/layer.js:95:5) at trim_prefix (/usr/lib/node_modules/signalk-server/node_modules/express/lib/router/index.js:328:13)`

miggat, 2024-03-10:
I also found this error:
`
Mar 10 12:17:15 (node:5040) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 delta listeners added to [FullSignalK]. Use emitter.setMaxListeners() to increase limit (Use `node --trace-warnings ...` to show where the warning was created)`

tkurki, 2024-03-10:
What is your Node version? `node -v`

miggat, 2024-03-11:
My node versión is 20.11.1
Maybe i should downgrade to v18?

tkurki, 2024-03-11:
No need for downgrade, I am barking up the wrong tree, the "not found" error is unrelated. That is probably triggered by a device on your network that is trying to make an access request.

miggat, 2024-03-11:
[image]

It does not seem theres any missing request pending.

miggat, 2024-03-11:
Is this connection setting correct?

[image]

miggat, 2024-03-11:
I don't think is related, but also found this error in SignalK l¡server log:
`
Mar 11 12:07:12 (node:5275) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 delta listeners added to [FullSignalK]. Use emitter.setMaxListeners() to increase limit (Use `node --trace-warnings ...` to show where the warning was created)`

miggat, 2024-03-15:
I really appreaciate your time helping me and I'm sorry for the time lost. After all it seems the problem is related to the NMEA2000 network wiring. Nothing to do with SignalK or the plugin.

closed 2024-03-15 by miggat