canboat/canboatjs #274: How to set the nmea200 device name

DrStS, 2024-06-21:
Hello,
I am using canboatjs with signalk. I noticed that the canboatjs device at the nmea network is shown as unkown from all the chart plotters in the network.
Is it possible to set the device name etc. to give the canboatjs device a proper name in the network?

Thanks a lot!
Best Stefan

macjl, 2024-06-21:
Hello,
I also noticed strange behavior with SignalK connected on N2k network. Sometimes, the device name appears, sometimes not. 
I suspect the PGNs 60928 and 126996 are not sent periodically, but only on demand from other devices. That’s why I can sometimes have the name of SignalK, when a device request it, but not every time.

macjl, 2024-06-21:
I supposed it was related to PGN 60928 and 126996, because the device name was into it. But it may be related to the heartbeat PGN. 

I din’t have a MFD, but I can see on Victron CerboGX that the SignalK device is not shown, unless I force the discover of NMEA devices. It’s not necessary for others N2k devices, that are always shown, such as YachtD gateway, LCJ anemometer, and Simrad GNSS.

DrStS, 2024-06-21:
I am using the Garmin  GPSMAP series 923 and 1623. I have updated to the most recent release. the issue is still there... looks to me that no device name is set. I have replicated that with a small eps32 example. If you do not set the device name Garmin reports unknown. Please let me know if I should provide more info. Thanks!

DrStS, 2024-06-25:
May I kindly ask if someone has an idea what I can try? Thanks!

sbender9, 2024-08-15:
I've just release v2.9.0 which now sends out heartbeats (pgn 126993). Let's see if that makes a difference?

sbender9, 2024-08-24:
Closing because of inactivity

closed 2024-08-24 by sbender9