canboat/canboatjs #286: Rusoku TouCAN on Raspberry Pi no longer works in Signalk (v 2.11.0) due to error: Cannot find module 'socketcan'

peterfouche, 2024-11-01:
Initial problem found in SignalK and replicated on the console by
pi@Siobhan-pi:~ $ candumpjs can0

node:internal/modules/cjs/loader:1143
  throw err;
  ^

Error: Cannot find module 'socketcan'
Require stack:
- /usr/local/lib/node_modules/@canboat/canboatjs/bin/candumpjs
    at Module._resolveFilename (node:internal/modules/cjs/loader:1140:15)
    at Module._load (node:internal/modules/cjs/loader:981:27)
    at Module.require (node:internal/modules/cjs/loader:1231:19)
    at require (node:internal/modules/helpers:177:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/@canboat/canboatjs/bin/candumpjs:6:19)
    at Module._compile (node:internal/modules/cjs/loader:1364:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1422:10)
    at Module.load (node:internal/modules/cjs/loader:1203:32)
    at Module._load (node:internal/modules/cjs/loader:1019:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:128:12) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ '/usr/local/lib/node_modules/@canboat/canboatjs/bin/candumpjs' ]
}

sbender9, 2024-11-01:
Something has gone wrong with your install. At install time, it was unable to build the socketcan module. 

I would start by reinstalling canboatjs and check the install logs to see why it failed.

peterfouche, 2024-11-01:
I did get the following when installing, but no other errors.

pi@Siobhan-pi:~ $ sudo npm install -g @canboat/canboatjs
npm warn deprecated inflight@1.0.6: This module is not supported, and leaks memory. Do not use it. Check out lru-cache if you want a good and tested way to coalesce async requests by a key value, which is much more comprehensive and powerful.
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported

added 113 packages in 2m

peterfouche, 2024-11-01:
so far it looks like python version issue. 

pi@Siobhan-pi:~ $ sudo npm install -g socketcan
npm error code 1
npm error path /usr/local/lib/node_modules/socketcan
npm error command failed
npm error command sh -c node-gyp rebuild
npm error gyp info it worked if it ends with ok
npm error gyp info using node-gyp@10.2.0
npm error gyp info using node@18.20.4 | linux | arm
npm error gyp info find Python using Python version 3.7.3 found at "/usr/bin/python3"
npm error gyp info spawn /usr/bin/python3

snip

npm error   File "/usr/local/lib/node_modules/socketcan/node_modules/node-gyp/gyp/pylib/gyp/common.py", line 435
npm error     if CC := os.environ.get("CC_target") or os.environ.get("CC"):
npm error            ^
npm error SyntaxError: invalid syntax

Seems a minimun of python 3.8 is needed to get a good compile? Do I have that right?

sbender9, 2024-11-21:
Did you get this resolved?

peterfouche, 2024-11-21:
Oh yes. Sorry I forgot to update. The issue was that canboat/canboatjs will
not compile anymore on a Raspberry Pi running Debian 10 (Buster) because
Python is at version 3.7. A minimum of Python 3.8 is needed to
successfully compile canboat/canboatjs. The solution for me was to rebuild
the whole Pi to the latest OS running Debian version: 12 (bookworm).
Anyone on an old Pi should avoid upgrading SignalK until they are ready to
upgrade the Pi itself I would think.

peterfouche, 2024-11-21:
Oh yes. Sorry I forgot to update. The issue was that canboat/canboatjs will
not compile anymore on a Raspberry Pi running Debian 10 (Buster) because
Python is at version 3.7. A minimum of Python 3.8 is needed to
successfully compile canboat/canboatjs. The solution for me was to rebuild
the whole Pi to the latest OS running Debian version 12 (bookworm).
Anyone on an old Pi should avoid upgrading SignalK until they are ready to
upgrade the Pi itself I would think.

closed 2024-11-21 by peterfouche