canboat/canboatjs #290: Error parsing 127489 Error: Cannot get 1 bit(s) from offset 56, 0 available

negrusti, 2025-01-06:
Can provide a log in PM
```
Error parsing 127489 Error: Cannot get 1 bit(s) from offset 56, 0 available
Error: Cannot get 1 bit(s) from offset 56, 0 available
    at BitView.getBits (C:\Users\User\AppData\Roaming\npm\node_modules\@canboat\canboatjs\node_modules\bit-buffer\bit-buffer.js:53:9)
    at BitStream.readBits (C:\Users\User\AppData\Roaming\npm\node_modules\@canboat\canboatjs\node_modules\bit-buffer\bit-buffer.js:365:23)
    at fieldTypeReaders.BITLOOKUP (C:\Users\User\AppData\Roaming\npm\node_modules\@canboat\canboatjs\lib\fromPgn.js:911:13)
    at readField (C:\Users\User\AppData\Roaming\npm\node_modules\@canboat\canboatjs\lib\fromPgn.js:588:13)
    at Parser._parse (C:\Users\User\AppData\Roaming\npm\node_modules\@canboat\canboatjs\lib\fromPgn.js:214:21)
    at Parser.parseString (C:\Users\User\AppData\Roaming\npm\node_modules\@canboat\canboatjs\lib\fromPgn.js:472:24)
    at Interface.<anonymous> (C:\Users\User\AppData\Roaming\npm\node_modules\@canboat\canboatjs\bin\analyzerjs:58:12)
    at Interface.emit (node:events:517:28)
    at [_onLine] [as _onLine] (node:internal/readline/interface:423:12)
    at [_normalWrite] [as _normalWrite] (node:internal/readline/interface:617:22)
```

sbender9, 2025-01-09:
Please provide log data. You can PM on the Signal K Discord, or send via email. Or just paste the log entry here.

sbender9, 2025-06-26:
Closing, for inactivity