canboat/canboatjs #297: PGN 126996 not sent on CAN bus despite correct NMEA2000 plugin configuration

Machansen, 2025-05-07:
### Problem:
SignalK does not send PGN 126996 (Product Information) via the NMEA2000 plugin. The Model ID does not appear on the CAN bus, although the `signalk-to-nmea2000` plugin is properly configured.

### Expected behavior:
A CAN node with a defined model ID should broadcast PGN 126996 after startup.

### Test environment:
- Raspberry Pi 4B  
- OS: AVNav image (multiple versions: 2023, 2024, armhf and arm64)  
- SignalK versions: tested with 2.10.0 and 2.12.0  
- Using standard setup and installation via Raspberry Pi Imager  

### Plugin config (from settings.json):
```json
{
  "pipeElements": [
    {
      "type": "providers/simple",
      "options": {
        "logging": true,
        "type": "NMEA2000",
        "subOptions": {
          "type": "canbus-canboatjs",
          "interface": "can0",
          "uniqueNumber": 123456,
          "useCanName": true,
          "mfgCode": 999,
          "filtersEnabled": false,
          "filters": [
            {
              "source": "",
              "pgn": ""
            }
          ]
        }
      }
    }
  ],
  "id": "CANBOAT_NMEA2000",
  "enabled": true
}

Additional notes:
The signalk-to-nmea2000 plugin is enabled and functioning (e.g. 130312 data is sent).

CAN bus works (confirmed with other NMEA2000 devices).

PGN 126996 is never seen on the bus.

Tested with cansend – PGN can be injected manually.

Question:
Is PGN 126996 supposed to be sent automatically by canboatjs, and under which conditions? Is there a specific configuration or message exchange required to trigger this?

Thanks for your help!

Machansen, 2025-05-07:
### Additional test result (08.05.2025):

I connected a different NMEA2000 device to the same CAN bus setup — PGN 126996 appeared immediately on the bus.

This confirms that the CAN bus is working correctly and the issue is specific to the SignalK/canboatjs node.

Also tested multiple AVNav images (2023–2024, armhf/arm64), with both SignalK 2.10.0 and 2.12.0.

→ PGN 126996 is still never sent automatically.

→ Manually injecting PGN 126996 using `cansend` works — the message appears correctly on the bus.

Thanks again for looking into this!

sbender9, 2025-05-07:
This will only happen when Signal K starts up or another device asks for it. Are looking for it at startup?

sbender9, 2025-05-07:
Hmm. Actually, I only send this if another device asks for it.

Machansen, 2025-05-07:
Thanks for the clarification.

Confirmed: PGN 126996 is **not broadcast automatically** by canboatjs at startup.

→ It is only sent when another NMEA2000 device sends a PGN 59904 request.

This explains the behavior: in a test setup with only SignalK on the bus, PGN 126996 does not appear — which is expected.

Issue can be considered clarified.

closed 2025-05-08 by sbender9